NorthernWidget/DS3231 #7: Alarm 2 gives wrong Alarm bits back

SimGas, 2017-11-30:
Hi,

I discovered that Alarm 2 gives wrong alarm bits back when reading them. I'm not sure if the wrong bits are written to the controller or if just the reading is wrong. but I think its a combination of both problems.
Alarm 1 works totally fine

tichachm, 2018-01-05:
Hi,
I also have a problem with Alarm 2. I would like to run the alarms every day (hour, minute match), but Alarm 2 is triggered every hour (minute match).

Here is my code adding the Alarm:
Clock.setA2Time(0, 14, 02, 100, true, false, false);

Alarm 1 is working fine.
Thanks

SimGas, 2018-01-05:
Have you tried reading the Alarm2 bits? Do they show the same value as you gave them(100)?

tichachm, 2018-01-06:
Yes, the bits returned show 96

SimGas, 2018-01-06:
Can you try the following:

Clock.setA2Time(0, 14, 02, 0b1000000, true, false, false);

In my code I use sucessfully the following:

  rtc.setA1Time(0, 0, 0, 10,0b00001110, false, false, false); // 1110 means: Alarm, when seconds(10) match
  rtc.setA2Time(0, 0, 0, 0b1110000, false, false, false); //111 means: Alarm every minute, at second 00

tichachm, 2018-01-07:
Hi SimGas,
I changed the value to 0b1000000 it seems to work fine, thanks for your help.

awickert, 2021-02-12:
Thanks, folks, for helping each other!

closed 2021-02-12 by awickert