NorthernWidget/DS3231 #11: Century "untouchable"

jerome13013, 2019-01-10:
Hi,

Within DS3231.cpp, there is no possibility to update the "century" bit at adress 0x05h, bit 7.
You can read it with getMonth(&century) but using decToBcd(month) in the setMonth(month) routine doesn't allow to write it with for example setMonth( 0x80h OR month ).

A standalone function setCentury(bool Century) could be useful to complete the library :)

Yes... I know... the "century" bit is not very useful ;)

Best regards

IowaDave, 2022-08-16:
The datasheet says very little about the Century bit, only that, "The century bit (bit 7 of the month register) is toggled when the years register overflows from 99 to 00."

There is no mention that the bit plays any role in managing other time or date registers within the DS3231.

It might have only informational value. In that case, the bit might be just as usefully maintained in a program variable, or in EEPROM.

IowaDave, 2022-09-18:
@awickert @jerome13013 I propose closing this Issue as "wontfix" for the following reason:

Even though the Century bit is technically writeable, no function of the DS3231 is affected by writing any value to it. The bit operates to provide information in one direction only: to the user. Its value is potentially highly significant. It signals that the 8-bit value in the "year" register, 0x06, has rolled over from 99 to 00. Perhaps that information should not be easy to erase?