NorthernWidget/DS3231 #17: INT/SQW pin stays LOW following triggered alarm

adgsm, 2019-09-09:
Once alarm is triggered method [DS3231::turnOffAlarm(byte Alarm)](https://github.com/NorthernWidget/DS3231/blob/master/DS3231.cpp#L570) does not reset A1F / A2F bits in status register 0x0F.  This makes SQW pin stay LOW and thus making following interrupts impossible.

Simple solution would be to add following codes at the end of the method to modify status byte as well.

```
	// clear A1F/A2F flags for previously set alarms
	temp_buffer = readControlByte(1);
	// modify status byte
	if (Alarm == 1) {
		temp_buffer = temp_buffer & 0b11111110;
	} else {
		temp_buffer = temp_buffer & 0b11111101;
	}
	writeControlByte(temp_buffer, 1);
```

Thanks for the library!

flostuerzer, 2020-05-28:
Good evening Sir,

I am having the exact same issue. After the first interrupt, the SQW Pin stays low. I have tried to add your lines to the existing method in the .cpp file, but it still is not running as I want. Do you might have any other idea being the reason for this cas? Or is something else to do?

Thank you very much in advance.

Flo

Juan27fp, 2020-06-28:
Hello.
Could anyone tell how to activate the SQW pin to activate an externan interrupt on any pin of my micro?
I have already done the activation of the alarm2 and i can read the state with Clock.checkIfAlarm(), but i'm not able to control the SQW pin in order to activate an interruption.
Thanks in advance!

Juan27fp, 2020-06-28:
> Good evening Sir,
>
> I am having the exact same issue. After the first interrupt, the SQW Pin stays low. I have tried to add your lines to the existing method in the .cpp file, but it still is not [...]

Have you tried using the checkIfAlarm function? once it reads the alarm on, it also resets it

jnuernberg, 2021-02-26:
Hello, 
I believe this issue boils down to a "How to use this library"-kind of situation. 

The functions `DS3231::turnOffAlarm()` and `DS3231::turnOnAlarm()` only affect the INTCN, A2IE and A1IE bits of the control register. A2IE and A1IE one enable the alarm wherease INTCN enables the output over the #INT/SQW pin. So if, for example, A2IE is set to logic one and the conditions match, this causes the alarm 2 flag (A2F) in the status register to go to logic 1. If additionally, INTCN is logic 1, this also causes the #INT/SQW pin to go low. 

After an alarm has been fired, the alarm flags (A1F and A2F) are logic 1 and have to be actively reset to logic 0 to allow for the next alarm. So to say, each alarm has to be acknowledged. This is correctly done by writing logic 0 to the alarm flags A1F and A2F 
in the status register (the code snippet from the original post does this properly). In this library, this is implemented in the `DS3231::checkIfAlarm()` method. The #INT/SQW pin is following the action of the alarm flags.

So the following pseudo-code should correctly represent usage of this library:
```
  // ===== Setup =====
  // Set alarm to repeat every full minute
  Clock.setA1Time(0, 0, 0, 0, 0b00001110, false, false, false);
  Clock.turnOnAlarm(1);

  // ===== Polling =====
  while (True){
    // check if alarm fired, and acknowledge if alarm fired
    bool alarmed = Clock.checkIfAlarm(1);
    // alarm action
    if (alarmed) {
      // alarm response
    }
  }

  // ===== Interrupt based =====
  // Interrupt occurs 
  // alarm response
  // acknowldge alarm
  Clock.checkIfAlarm(1);
  // Wait for next alarm interrupt

  // ===== Clean up =====
  // Turn alarm of, if no longer needed.
  Clock.turnOffAlarm(1);
```
I suggest to acknowledge the alarm only after a proper alarm response, such that one alarm cannot interrupt another alarm response in the interrupt based scheme. 

I hope this helps and clarifies usage.
Cheers, Jacob

Manujuggler, 2021-02-26:
@jnuernberg

I created an account to thank you for your post. I was stuck and this helped a lot ! 
When comparing to other librairies as well as chip datasheet, I thought the flag was not acknowledged correctly. It saved me time.
Again, Thanks Jacob.

jnuernberg, 2021-02-26:
I created my account just to mention this ;) So, definitely worth it. Happy coding.

adgsm, 2021-02-26:
@jnuernberg explained well and provided a solution

closed 2021-02-26 by adgsm

jnuernberg, 2021-03-01:
Short post note, to who is still having trouble here:
I had some trouble debugging, when I interrupted my code before calling ´CheckIfAlarm()´. It was then stuck with the INT/SQW pin low. It helped to completely power down the DS3231 for a few seconds and then run again (remove battery as well).

AlohaCPU, 2022-07-17:
I'm looking for basic information on the DS3231.h library, the above helps, but there are soo many more functions (https://github.com/NorthernWidget/DS3231) in this library that I cannot find straight forward how-to information and examples. 
In my situation I'm am using the basic code "https://github.com/ostaquet/arduino-nano-33-iot-ultimate-guide/blob/master/src/PowerOnTimer/PowerOnTimer.ino#L25" and circuit "https://github.com/ostaquet/Arduino-Nano-33-IoT-Ultimate-Guide/blob/master/images/WakeOnTime_schem.png". I've got the software and hardware to work, but I need to make changes to the alarm intervals and to understand the SQW pin change with respect to the function calls in the code.
Specific questions are:
(1) When the DS3231 is powered-up, what is the logic status of SQW, from my read it is HIGH?
(2) What is the logical status of SQW when the call to "Clock.turnOnAlarm(1);" executes, is it still HIGH?
(3) Why are the parameters to the call "Clock.setA1Time(1, 2, 3, 4, 0b1110, false, false, false);" set as is,
      * The first three parameters  "1,2,3" are not even used as I understand it, 
      * The fourth "4" means that the Alarm "1" is triggered the fourth second of each minute? 
      * The "false, false, false" indicates that since we are triggering the alarm every minute, the first/second/third parameters are ignore.
      * How do I set the alarm for every three minutes or every 45 seconds?
 (4) When "Clock.checkIfAlarm(1);" is executed,
        * SWQ goes HIGH or LOW
        * Does this call initiate the "one minute count down" until the Alarm "1" causes the SWQ to go from HIGH to LOW,
        * Or does it assume that Alarm "1" has already triggered SWQ to go from HIGH to LOW and now resets SQW to HIGH?
   I apologize for such basic questions, but the I really haven't found a good easy to understand resource for this library. Thank you for your help.

jnuernberg, 2022-07-20:
Hallo @AlohaCPU, 

I cannot test my suggestions right now as I don't have any hardware at hand, but let me answer to the best of my knowledge. [The datasheet](https://datasheets.maximintegrated.com/en/ds/DS3231.pdf) is your friend in this case, it contains almost all information, that you will need.
Provided, that your IDE uses the same Library as this one, this should help you:

1. The SQW-pin is active LOW. The datasheet states: "Because the INTCN bit is set to logic 1, when power is first applied, the pin defaults to an interrupt output with alarms disabled", meaning, that the SQW-pin should default to logic zero (i.e. HIGH) on startup (and not in square wave mode).
2. A call to `DS3231::turnOnAlarm(1)` should not affect the status of the SQW-pin, so (unless there was an unacknowledged alarm) the pin should remain HIGH.  This call only sets the INTCN, A1IE or A2IE registers to logic one. These enable the alarm (this is not a timer, but an alarm by bit comparisson).
3. The call signature is :
```void DS3231::setA1Time(byte A1Day, byte A1Hour, byte A1Minute, byte A1Second, byte AlarmBits, bool A1Dy, bool A1h12, bool A1PM)```
With `byte AlarmBits` set to `0b1110`, we set the alarm to go of, everytime the seconds that we set, match the current time. So you are right, the first three arguments are unused, likewise `bool A1Dy`. The last two arguments define the  time format, that we use to set the time. I suggest, that you read up in the datasheet on the AlarmBits (Page 12).
To set the alarm to go of every 3 minutes or every 45 seconds, you have to set the correct time after each call (as there is no simple method to match every 45 s or 3 min to a constant). For every 45 s, you can use the same AlarmBits are here (`0b1110`), for every three minutes, you need to match on the minute as well (`0b1100`). 
4. `DS3231::checkIfAlarm` resets the SQW pin to logic 0 (HIGH), indifferent of its current state. Again, this is not a countdown or timer, this is a bit match alarm, i.e. there is no "starting" of any kind. 

I hope this helps. Again, all this information can be found in the datasheet as well.

Best regards, 

Jacob

AlohaCPU, 2022-07-21:
Thank you this is great information. The terminology in the data sheet was confusing. Here is why I had asked. I breadboarded and coded the example: https://github.com/ostaquet/Arduino-Nano-33-IoT-Ultimate-Guide.
As you mentioned, SQW should start HIGH per the data sheet, and I have seen this stated elsewhere. But if that was so, on power-up the MOSFET in the schematic would be off and the Nano would not be powered, and thus not execute the code, but it does power-up in my testing, and indeed SQW starts in a LOW state?! This is why I’m confused.
Next the system stays powered up until DS3231::checkIfAlarm is called. At this point even though the match time hasn’t trigger Alarm 1 yet (this only apples on the first power-up sequence), SQW goes high and the system powers down. And I’ve now verified this also. When the Alarm 1 fires, SQW goes low and the power-up occurs and we repeat. O K, I can live with this discrepancy in the data sheet. I suspect that we can’t assume that because INTCN is in a HIGH state initially, we cannot assume the unstated initial states of SQW, A1IE/A2IE or A1F/A2F, at least I haven’t found it in the spec sheet.
My next problem is why I wanted to be able adjust the Alarm time. I’m running my code in the Arduino Cloud IOT environment. The start-up requires connecting to the WiFi then the cloud, this is often a 50 secound process, really slow. Since I’m trying to run off of batteries, a power ON:OFF of 5:1 isn’t much help. Hence I need to figure out how to pass appropriate time parameters to accomplish this. So setting 0b110 allows me to specify the next minute and seconds value that will trigger Alarm 1. I’ll get started!
Thanks again.

Aaron Chun