NorthernWidget/DS3231 #24: DateTime comparison operators

namanvs, 2020-06-15:
Can we get '<', '>', '-', '+' operator overloading for the DateTime class?

awickert, 2020-06-23:
Please do submit a PR! I am no longer very actively maintaining this code, so community support is much appreciated.

IowaDave, 2022-08-16:
Overloading might be desirable. If @namanvs  would submit a PR I would be happy to review it.

It may be worth noting, however, that such comparisons and arithmetic can be done presently without overloading. The approach would be to calculate a difference in seconds, taking advantage of the ```DateTime::unixtime()``` method.

Code segment:

```
DateTime A(); // assume initiated properly
DateTime B(); // assume initiated properly
uint32_t a = A.unixtime();
uint32_t b = B.unixtime

if (a > b) { // A is after B }

uint32_t diff = a - b;
```

awickert, 2022-08-19:
Good point @IowaDave.

I don't have need for this overloading so won't be dev'ing it. How about this as a plan?
* If we hear from @namanvs soon-ish about creating a PR, great
* Otherwise, we close this as `wontfix` with reason "not enough current motivation"

namanvs, 2022-08-20:
Hi @awickert and @IowaDave, sorry, I've been travelling but I would be happy to write these operators. I will need about 10 days to get back up to speed and have something ready, but I plan on implementing the following operators: '<', '>', '+', and '-'. I'll probably use @awickert 's suggested strategy but wrap it up into the operators for convenience. I am toying with a modulo operator to return the remainder between 2 operands in the next-smallest unit:

DateTime a % DateTime::DAYS returns number of hours in the remainder.

Would this seem useful?

IowaDave, 2022-08-20:
I will wait to complete and PR my documentation on DateTime until after the
proposed enhancements to DateTime are in place.

Thanks!

IowaDave, 2022-09-15:
@namanvs , @awickert , I have added the DateTime documentation to the master branch of this repo. In the near future, PR #70 may add an example program that uses a DateTime calculation to advance the time values of an alarm.

I'll look forward to viewing that PR you mentioned, if you choose to offer it.

Thank you sincerely,

David

namanvs, 2022-09-16:
I have submitted a PR. This is actually the first time I'm contributing to code on Github. Open to feedback and willing to make improvements where identified.

IowaDave, 2022-09-16:
@namanvs Welcome! I look forward to reviewing your code. 

Please see my comments in the PR (#71). Thanks!

IowaDave, 2022-09-18:
Changed the label to Enhancement because that is really what @namanvs is proposing. Leaving this Issue open pending resolution of his PR #71.

cc: @awickert