NorthernWidget/DS3231 #25: uninitialized const 'buff' [-fpermissive]

lbrutti, 2020-07-20:
Hello,
tried to compile a sketch for MKRFOX1200 and got this error:
`/Users/xxx/Documents/Arduino/libraries/DS3231-master/DS3231.cpp: In constructor 'DateTime::DateTime(const char*, const char*)'`
`/Users/xxx/Documents/Arduino/libraries/DS3231-master/DS3231.cpp:130:22: error: uninitialized const 'buff' [-fpermissive]
    static const char buff[4];`

I added the initialization code and everything was fixed, please consider my PR ;)

hasenradball, 2020-08-01:
Hi replace the following line 130 with:
` static const char buff[4] {};`
The It should fixed!

hasenradball, 2020-08-01:
Where is you PR?

hasenradball, 2020-08-02:
see PR #28

lbrutti, 2020-08-02:
youre right, just submitted it 👍

hasenradball, 2020-08-02:
I also submitted one on the morning, lets see which one wins 😀

awickert, 2021-02-12:
A whole train of snails passed you folks, but we're finally there!

closed 2021-02-12 by awickert

hasenradball, 2021-02-12:
hi 

what is the mesning of your comment?

Frank

lbrutti, 2021-02-12:
Thanks for merging!

🙂

Have a nice day

awickert, 2021-02-12:
> hi 
> 
> what is the mesning of your comment?
> 
> 
> Frank
> 
> 
> > 
> > 
> > A whole train of snails passed you folks, but we're finally there!

Dass habe ich endlich der PR finalisiert :)