NorthernWidget/DS3231 #29: RTC freezing and locking up

marcelocsi, 2020-10-02:
Battery powered RTC seems to be frozen after Arduino power up. The command RTC.now() is reporting 165:165:85 which does not make any sense. As the device is powered and battery is soldered, I am unable to cut its power.

Proposed solution: the class should provide a mechanism to control manually the logic level of the output digital pins to attempt to unlock the communication channel, e.g. the i2c bus is always "busy".

awickert, 2021-02-12:
This sounds like the digital logic is clamped to some HIGH or LOW value. If someone feels like taking this on, more power to you!

marcelocsi, 2021-02-12:
I figured it out. The issue was the interrupt management. I was using atomic block while managing algorithms and one of them was dealing with the library. This library seems to be triggered after user's call and the interrupts were disabled. The library was then feeding back spurious data.

I validated it by downloading only the library without any of my application and the problem was gone, so, the hardware and device were fine.

Thanks for the comment, I am gonna close this topic.

closed 2021-02-12 by marcelocsi

awickert, 2021-02-12:
Perfect; thank you for the explanation!