NorthernWidget/DS3231 #31: where are descriptions of the functions?

3Dietrich, 2020-10-18:
Sometimes I feel really stupid, like right now.. sorry, but I need to ask: 
You write a lot of possible [functions](https://github.com/NorthernWidget/DS3231#functions), but I can't find any description..?
I know I can open the examples and can find several functions but not all and not ordered.

awickert, 2021-02-12:
Yay open source!
If someone finds it worthwhile to document this library, please do.
I may do it, or find someone to do it, but it'll be on the timeline in which it is useful to my research team and myself... not trying to be difficult! Just very busy!

closed 2021-02-14 by awickert

3Dietrich, 2021-02-14:
oh.. closed? So you found someone who finds it worthwhile to document this library?
If not - why do you close it?

awickert, 2021-02-14:
I closed this because this is a fairly generic and obvious request, rather than for some specific feature, and is therefore not something for which I need to have an open Issue to know to do. This should be obvious to anyone else as well. Therefore, I can keep the Issues page clean for things that are more difficult (if perhaps not as time-consuming) to solve.

I would keep open a request for a specific piece of documentation.

3Dietrich, 2021-02-14:
ok, thanks for the explanation