NorthernWidget/DS3231 #33: add function documentation?

Pomax, 2021-01-23:
There does not appear to be any documentation for this library anywhere, which isn't great. There are examples, but that doesn't tell anyone what the function signatures are, what input they expect, and what output they yield. Can that be added? Also, as a splice of the Adafruit library, where are the `initialized` and `adjust` functions? Because those are _incredibly_ useful =)

(being able to say `rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));` makes setting the RTC entirely trivial compared to having to call seven separate functions)

awickert, 2021-02-12:
Adding documentation sounds like a great idea! This was a splice with a portion of the JeeLabs/Adafruit library MANY years ago... so I'm not sure that those other functions even existed.

If they're useful to you, I'd suggest you edit and make a PR!

Pomax, 2021-02-13:
I know nothing about this library, I'm just a user who hates hardware hacking, and wanted to find something that would work with the DS3213 I had to connect to an arduino to enable some functionality with minimal programming =)

I ran across this one, noticed it seemed under-documented, and figured I'd let you know in case you wanted to improve that, but if it's a very old fork of someone else's project, that's probably not worth the effort.

closed 2021-02-13 by Pomax

awickert, 2021-02-13:
Don't lose hope! It's not an old fork of someone else's project, and I might add this the next time that I need it. In the meantime, hoping that the examples can get you going or that you find a good alternative.

laurensV, 2023-01-04:
`rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));` would be great to have!