NorthernWidget/DS3231 #36: DS3231 library getTemperature() function

ScooterKraig, 2021-02-14:
Code using the DS3231 library was working until the library was updated.  Is there something missing in nanf("")?

Arduino: 1.8.13 (Linux), Board: "Arduino Uno"

/home/kraig/Arduino/libraries/DS3231/DS3231.cpp: In member function 'float DS3231::getTemperature()':
/home/kraig/Arduino/libraries/DS3231/DS3231.cpp:414:16: error: 'nanf' was not declared in this scope; did you mean 'tanf'?
  414 |     temp3231 = nanf(""); // Not a number
      |                ^~~~
      |                tanf
exit status 1
Error compiling for board Arduino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

awickert, 2021-02-14:
Thanks for the note! This was requested in a PR but I did not test it myself. I have just returned the error temperature to -9999 (d99c91faef060cd21c815732b0060f3bcb04b755), as it was before, and pushed a new incremented release (v1.0.7) to be picked up by Arduino. Thank you for the note!

Closing; please reopen if this doesn't fix it and I will take a bit of time to debug.

closed 2021-02-14 by awickert

ScooterKraig, 2021-02-14:
Thanks.  That apparently solved the fnan() error.  Now I'm getting the
following with the DS3231_set and DS3231_test examples.  Both of these
compiled before the library was updated.
Arduino IDE 1.8.13.

Kraig Jones

/usr/bin/avr-ld: error while loading shared libraries: libctf.so.0: cannot
open shared object file: No such file or directory
collect2: error: ld returned 127 exit status
exit status 1
Error compiling for board Arduino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ScooterKraig, 2021-02-14:
Please disregard the previous message about compile errors.  Reloading
Arduino IDE cured that.

Kraig Jones