NorthernWidget/DS3231 #37: [New Idea]: set rtc by feed the epoch

hasenradball, 2021-03-14:
My idea is to make a method which has a parameter epoch which sets then the rtc, like time and date, automatically?
How do you think about?

awickert, 2021-03-15:
Sounds like a good idea! Indeed, I have thought about doing that too, but I haven't. It would just take writing some code to convert the epoch to the date/time parameters for the registers.

The reason that I have not yet done this is because I have written code that takes the computer time and directly sends it to the logger, thus negating the need for any human input. This link works with our older-model data loggers, but should be able to be ported to work with more generic Arduino devices: https://github.com/NorthernWidget/ALogTalk.

hasenradball, 2021-03-15:
I found this for the Arduino nano 33 iot. There you could set the time by giving the epoch.

It s a nice solution, but the RTC of the nano is a really bad rtc compared to the DS3231.

One point would be setting the correct local time.

My first proposal would be starting with 

`setEpoch(time_t epoch)`

and epoch in the first approach is the already corrected time value for the local time...

I can test if you want!

best regards

Frank

awickert, 2021-03-16:
If you would like to test some code and make a PR, please do! I do not have time or plans to make any significant edits here in the foreseeable future.

hasenradball, 2021-03-17:
Hi, 
I have one question....
I found the following constuctor:
```
DateTime::DateTime (uint32_t t) {
  t -= SECONDS_FROM_1970_TO_2000;    // bring to 2000 timestamp from 1970

    ss = t % 60;
    t /= 60;
    mm = t % 60;
    t /= 60;
    hh = t % 24;
    uint16_t days = t / 24;
    uint8_t leap;
    for (yOff = 0; ; ++yOff) {
        leap = isleapYear(yOff);
        if (days < 365 + leap)
            break;
        days -= 365 + leap;
    }
    for (m = 1; ; ++m) {
        uint8_t daysPerMonth = pgm_read_byte(daysInMonth + m - 1);
        if (leap && m == 2)
            ++daysPerMonth;
        if (days < daysPerMonth)
            break;
        days -= daysPerMonth;
    }
    d = days + 1;
}
```
It seems you can gibe the epoch, right?
But the RTC is then not feed with the values?
Can you give some informations?

Frank

hasenradball, 2021-03-17:
No I think it is not the RTC class...

awickert, 2021-03-21:
You are correct: this is in a separate class.

awickert, 2021-03-21:
Thank you for the PR https://github.com/NorthernWidget/DS3231/pull/38. Looks like you tested it too and it works, so I'm closing this.

closed 2021-03-21 by awickert