NorthernWidget/DS3231 #42: No Alarm Documentation

owenwastaken, 2021-04-08:
I am working with DS3231's alarm programming and there is no documentation. Can someone explain what these parameters mean?

`setA1Time(byte A1Day, byte A1Hour, byte A1Minute, byte A1Second, byte AlarmBits, bool A1Dy, bool A1h12, bool A1PM);`

awickert, 2022-07-26:
Thanks @IowaDave and @jnuernberg! I'm very short on time to maintain and document this repository these days, but if you find it useful, would it be helpful for me to give you write access here so you can include your documentation right with the library itself?

IowaDave, 2022-07-26:
Happy to help.

How would you like me to submit content? Git push followed by a PR? Or just
write directly to the repo?

I imagine the material could benefit from a rewrite to focus on the library
and the datasheet for the part. Your thought?

Appreciatively,

David

jnuernberg, 2022-07-27:
Hi @awickert, 
Thank you! Happy to help myself. 

Time for me as well is the limiting factor. I will opt for a fork and PRs, when I have something to contribute. Until then, I'll help out on the comments and issues whenever I find my 5 minutes of time. 

@IowaDave I will of course support you, in case you decide to opt for a rewrite. 

Jacob

awickert, 2022-07-27:
Hi @IowaDave David and @jnuernberg Jacob,

Thank you very much for your willingness to help document this repo, and I am glad that it's been helpful enough for you that you feel it would be worthwhile.

The current `README` is quite minimalist. I trust that whatever you create would be better! This library is a community resource, and was something that I modified from a professor at Chico State's code around a decade ago. So you should also feel free to make it fully your own.

I'm the one who is appreciative here! Expanding this library and seeing all those who have used it has been fantastic, but more than I can manage to support. Thank *you*.

IowaDave, 2022-07-27:
We're on it!

Understood that the README is home base for the documentation. Will
approach it that way.

awickert, 2022-07-27:
Thank you.

And PR if you want to prevent clashes, but otherwise, feel free to edit directly.

And just do what makes the most sense to you; it probably will to a bunch of other folk too.

IowaDave, 2022-08-06:
Hi there. Some PRs pending. Do you want to be the Approver on them? Or do
you want Jacob and me to go ahead and merge our changes?

Thanks!

awickert, 2022-08-08:
I'm very happy for the two of you to manage the PRs and take full control. In fact, this is my preference -- I'd rather not feel the possibility of slowing you down! But since I'm here anyway right now, I'll have a quick look :)

awickert, 2022-08-08:
Merged @IowaDave's docs, but keeping this Issue open because these don't yet include the answer to @owenwastaken's question.

IowaDave, 2022-08-08:
Very well. Jacob and I will exercise the discretion to merge. We'll monitor
it for PRs from other contributors also. To be clear, it's still your repo,
and we're grateful to you for providing it to the community.

David

awickert, 2022-08-09:
Thank you, but it is in fact simply my modification and splice of work by Eric Ayars and @jeelabs, and is licensed totally without conditions: https://github.com/NorthernWidget/DS3231/blob/master/LICENSE

So it is all of ours, and living here per history and convenience of location :)

IowaDave, 2022-09-15:
PR #70 adds extensive documentation for the Alarms. It is open for review by developers as of mid-September 2022. Hope to merge it soon. Thanks for your patience!

IowaDave, 2022-09-18:
Closing this Issue on 18Sept2022 because alarm documentation has been added to the repository. See [Alarms](https://github.com/NorthernWidget/DS3231/blob/master/Documentation/Alarms.md).

closed 2022-09-18 by IowaDave