NorthernWidget/DS3231 #45: [Question]: get Epoch value from DS3231 possible?

hasenradball, 2021-05-23:
Hi Andy,

I introduced a method 'setEpoch' to the DS3231 code.
Now I am thinking about to query the Epoch value back from the Ds3231 to feed the (for example) every hour the ESP8266.

Background is that the ESP8266 has ab very bad time accuracy, and want to feed it every hour by the Epoch queried from the RTC.

Is it possible to get the Epoch from the DS3231?

awickert, 2021-05-24:
It is, via `unixtime()`.

hasenradball, 2021-05-24:
hi Andy oh I see there is one method.
😉

Do you know what ist the empty CTOR for in the 
DS3231. cpp file?

Should be deleted the compiler does this job for you.

awickert, 2021-07-03:
Thanks for the thought! I'd test this first; early Arduino, at least, required such empty constructors, and I haven't taken the time to optimize for any potential compiler changes.

Closing this thread as it's now on a bit of a different topic.

closed 2021-07-03 by awickert

RobS2332, 2024-06-29:
It does not seem to handle ESP32Time epoch values. Why is this?

hasenradball, 2024-06-29:
@RobS2332 
Hi have you an short example for the issue?

hasenradball, 2024-06-29:
@RobS2332 
I checked „ESP32Time“ is a Arduino IDE Library.

What is supported is the cpp official „time.h“ library.
[time.h](https://cplusplus.com/reference/ctime/)

RobS2332, 2024-06-29:
The  standard is to call WiFi then the new ESP32Time library for the unix values. See below. rtxExt would be the DS3231, rtcOb the  devkit ONBOARD RCT.

    /*---------set with NTP using struct---------------*/
  struct tm timeinfo;
  while (!getLocalTime(&timeinfo)) {
    configTime(gmtOffset_sec, daylightOffset_sec, ntpServer1,ntpServer2);
    struct tm timeinfo;
    if (getLocalTime(&timeinfo)) {
        esp32Time.setTimeStruct(timeinfo); // Set time to ESP32Time instance
    }
  }
    rtcOB.setTimeStruct(timeinfo);
    // Get the epoch time from esp32Time
    uint32_t ntpTime = esp32Time.getEpoch();
    // Set RTC to esp32 Epoch Time
    rtcExt.adjust(DateTime(ntpTime));

You split it out to the RTCLib but implement your own Clock. this gives me a year of 1984 consistently! 

I'm trying to find something smaller than Adafruit for use on a ESP32-C3

hasenradball, 2024-06-30:
Hi,

did you had a look to the DS3231 Datasheet?
And checked how different the datasheets are?

Can you please format the code above as code so it is better readable.
Thanks.

hasenradball, 2024-06-30:
The additional question is.
When you have wifi active why don‘t you feed directly the unix timestamp into DS3231 as epoch.