NorthernWidget/DS3231 #48: Update to 1.10. brocke compling, but it work with old library V1.0.7  !

Jens869, 2021-12-07:
Arduino: 1.8.17 Hourly Build 2021/09/06 02:34 (Mac OS X), Board: "ATmega2560, Yes (UART0), EEPROM retained, Arduino MEGA pinout, BOD 2.7V, LTO enabled, External 16 MHz"

/Users/name/Documents/Arduino/libraries/DS3231/DS3231.cpp: In constructor 'DateTime::DateTime(uint32_t)':
/Users/name/Documents/Arduino/libraries/DS3231/DS3231.cpp:104:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (days < 365 + leap)
             ~~~~~^~~~~~~~~~~~
/Users/name/Documents/Arduino/libraries/DS3231/DS3231.cpp: In member function 'void DS3231::setEpoch(time_t, bool)':
/Users/name/Documents/Arduino/libraries/DS3231/DS3231.cpp:264:12: error: aggregate 'DS3231::setEpoch(time_t, bool)::tm tmnow' has incomplete type and cannot be defined
  struct tm tmnow;
            ^~~~~
/Users/name/Documents/Arduino/libraries/DS3231/DS3231.cpp:266:3: error: 'localtime_r' was not declared in this scope
   localtime_r(&epoch, &tmnow);
   ^~~~~~~~~~~
/Users/name/Documents/Arduino/libraries/DS3231/DS3231.cpp:269:3: error: 'gmtime_r' was not declared in this scope
   gmtime_r(&epoch, &tmnow);
   ^~~~~~~~
/Users/name/Documents/Arduino/libraries/DS3231/DS3231.cpp:269:3: note: suggested alternative: 'time_t'
   gmtime_r(&epoch, &tmnow);
   ^~~~~~~~
   time_t
Mehrere Bibliotheken wurden für "Timezone.h" gefunden
 Benutzt: /Users/name/Documents/Arduino/libraries/Timezone
 Nicht benutzt: /Users/name/Documents/Arduino/libraries/Timezone-master
Mehrere Bibliotheken wurden für "TimeLib.h" gefunden
 Benutzt: /Users/name/Documents/Arduino/libraries/Time-master
 Nicht benutzt: /Users/name/Documents/Arduino/libraries/Time
exit status 1
Fehler beim Kompilieren für das Board ATmega2560.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

awickert, 2021-12-09:
Schade.
Beim jetzt, bitte benützen Sie der v1.0.7.
Ich arbeite jetzt dieses Problem zu lösen.

designer2k2, 2021-12-10:
1.1.0 broke for me too, worked with 1.0.7, and now:
`C:\Users\WIN-7\Documents\Arduino\libraries\DS3231/DS3231.h:22:18: fatal error: time.h: No such file or directory
 #include <time.h>
                  ^
compilation terminated.
`

im using it on a Digistump Pro  (Attiny167)

Jens869, 2021-12-12:
May be a problem with time.h and DS3231.h ?
I want use an GPS clock too. And that is always UTC = GMT. But I need MET/MEST (local Time +1/+2)

awickert, 2021-12-13:
Thanks, both @Jens869 and @designer2k2: I am wondering if one of the PRs may have inserted `time.h` as a new dependency, which must help somehow, but breaks support otherwise.

@Jens869: You are able to return local time as well using this library. Also (importantly) UTC != GMT during summer time.

awickert, 2021-12-13:
I found the source of the problem, and it was a change that I made: https://github.com/NorthernWidget/DS3231/commit/2e7d0323acf8466442e6f6b7362f8076675e4a47

Commit message:
```
#include <ctime> to #include <time.h>: updated library name
```

So it must be that the library name was updated in Arduino, but not everywhere.

A solution could be to create conditional includes.

@Jens869: I will not be able to solve your problem until you share your code (and ideally, which microcontroller system you are using).

Jens869, 2021-12-13:
my Laptop is today and tomorrow  in repairand I have not allways Time for arduino but it make sencse, if the Rename is Not always the same.I want to read UTC from GPS Serial konvert it in Local time Mittel Europäische Zeit oder Sommerzeit put it in the DS3231 and read it and display itthen I Don t Need a Menu to set the clock Jens Schröder

awickert, 2022-01-06:
Final note and closing:

`ctime --> time.h`: In Arduino, but not everywhere.
Relevant diff: https://github.com/NorthernWidget/DS3231/commit/2e7d0323acf8466442e6f6b7362f8076675e4a47

Conditional #include would require more knowledge on the other systems and how to write the proper conditional.

closed 2022-01-06 by awickert