NorthernWidget/DS3231 #53: setEpoch() does not work with Unix Time but with time since 01/01/2000 00:00:00

bill-ba, 2022-03-09:
The documentation indicates to use Unix time.
Trying to set DS3231 RTC using setEpoch() function gives wrong dates.
Substracting 946684800ul (30 years) from 'epoch' parameter solves the issue.

leslieadams57, 2022-04-05:
I've found the same problem when setting epoch (seconds since 1900 -70 years) and then re-read the year I'm getting 1992 instead of 2022. Having looked at DS3231.h there is mention of times since 1/1/2000 and 1/1/1970 but was unsure how this fitted.
My work around is to set epoch and then adjust the year by 30. Any ideas???

bill-ba, 2022-04-05:
setEpoch() uses gmtime_r() which expects time elapsed since 1/1/2000.
As written above, a way to calculate the good parameter is : unix_time = y2k_time + UNIX_OFFSET; // #define UNIX_OFFSET 946684800UL in source code
So : y2k_time = unix_time - UNIX_OFFSET;

getYear() returns a byte (8 bits, value 0-255), not an integer ; the returned value should be treated as a byte or properly casted. It's weird to get 4 digits...

leslieadams57, 2022-04-05:
bill-ba cheers !!! I'd guessed that setEpoch() was working to 1/1/2000 as it was always 30 years out. I'm getting current time from an NTP server (e.g. days from 1900) and was converting to 1/1/1970 e.g. Unix which I thought that setEpoch() would use. Just changed the code to reset from 1970 to 2000 (e.g. +30 years) and all OK. However when I use setEpoch() and then re-read the year, it comes back as 4 digits e.g. 2022. I have the year defined in getYear() as "uint16_t year;" in a structure (i.e. not 8bits but 16) and it seems to work OK!

bill-ba, 2022-04-06:
Hey
you can also use the following macro :
#define NTP_OFFSET 3155673600 // already in source code too
So : y2k_time = _1900_time - NTP_OFFSET;

The arduino DS3231.h contains the function 
byte getYear(); 	// Last 2 digits only 
in class DS3231

leslieadams57, 2022-04-06:
Hi, yep your quite right, sorry it's my mistake. I'd changed from DS3231.h to RTC.h which defines a DS3231 class but with "uint16_t getYear(); " instead. This is what is returning the 4 digit year. RTC.h caters for DS3231, DS1307 & PCF8563 so you can declare whichever one you like. Cheers !!!

awickert, 2022-08-15:
Thanks for these comments! @bill-ba and @leslieadams57. I think I am understanding what is going on.

@hasenradball added this functionality: see c5356092a911c9225b8a643ab054de0446c090b3.

When I merged the PR, I presumed that the epoch for `gmtime_r` and `localtime_r` was Jan 1, 1970. But it turns out that for the AVR implementation, it is indeed Jan 1st, 2000!
https://www.nongnu.org/avr-libc/user-manual/group__avr__time.html

This then led to some confusion between myself and @IowaDave.

My plan here is to correct these to use Jan 1, 1970 as the datum. Towards this end, I have just created the following PR: #65.

Hopefully we can move on resolving this Issue soon (and at last!)

hasenradball, 2022-08-16:
@bill-ba @leslieadams57 @awickert 
Hello guys,
thanks for you issue!
But wahat is no so optimal at this issue for my opinion is the details behind the scenes...

I am willing to solve the issue if there is on, but if you only say sonething is not going right thats too less I think.

Please consider the Lib is used on many microcontrollers,

The AVR platform incluces a differnt `time.h` than the ESP8266.
So If you only say `nothing goes right` I could only say `then go left` :-)

So for the future ... plese provide minimal the following things:

- a simple sketch to reproduce the issue
- which platform or microcontroller you are using
- and if possible additional info

awickert, 2022-08-16:
@hasenradball and all: Let's please move this discussion to my PR, linked above (#65). I will reply there.

IowaDave, 2022-09-05:
@bill-ba , @leslieadams57  Thank you for identifying this issue. The problem turned out to involve differences between the Arduino IDE Core library for AVR processors such as Uno and Nano, in contrast to the one for ESP8266. The old code worked for ESP8266 but not for AVR.

Closing the issue September 2022 after the setEpoch() function was updated and corrected for AVR processors by PR #66. It has been tested and found to work correctly now for both AVR and ESP8266 devices. Thanks to @awickert  and @hasenradball  for their work on this bugfix.  
- IowaDave

closed 2022-09-05 by IowaDave

hasenradball, 2022-09-28:
@IowaDave @awickert 
Hi together,

will you plan a new release with the latest changes?

Frank

IowaDave, 2022-09-29:
Soon I hope. We have an enhancement PR pending for some operator overloads
on the DateTime class that Andy asked to see. It might be that he would
choose to merge the PR before making a new release. I would like to get his
thoughts on this.