NorthernWidget/DS3231 #91: undefined reference to DateTime::dayOfTheWeek()

Danccce, 2023-09-16:
Hi everybody,

I'm using the DS3231 module and tried to modify the "now" example code:
```
void loop () {
    
    delay(1000);
    
    DateTime now = myRTC.now();
    
    Serial.print(now.year(), DEC);
    Serial.print('/');
    Serial.print(now.month(), DEC);
    Serial.print('/');
    Serial.print(now.day(), DEC);
    Serial.print(' ');
    Serial.print(now.hour(), DEC);
    Serial.print(':');
    Serial.print(now.minute(), DEC);
    Serial.print(':');
    Serial.print(now.second(), DEC);
    Serial.println();
    
    **Serial.print(now.dayOfTheWeek(), DEC);
    Serial.println();** 
```

but I got the error message
`.../now/now.ino:37: undefined reference to `DateTime::dayOfTheWeek() const`

what am I doing wrong? I use Arduno IDE 2.2.1 and DS33231 lib 1.1.2

Thank you
Marc

hasenradball, 2023-10-17:
@Danccce @IowaDave ,

I would say the DateTime Class has  a declaration of the `dayOfTheWeek()` function, but the function is not defined.
We need to define this function.

hasenradball, 2023-10-18:
@Danccce @IowaDave 
I prepared an solution.
Tested on ESP8266
See PR #94

hasenradball, 2023-10-27:
@Danccce for this Issue you can try this library [DS3231-RTC](https://github.com/hasenradball/DS3231-RTC).
It is also available in Arduino now.