NorthernWidget/DS3231 #102: How to format time by "hh:mm:ss"?

N3koSamuRay, 2025-05-11:
```
DateTime now = myRTC.now();
  int year = now.year();
  int month = now.month();
  int day = now.day();
  int hour = now.hour();
  int minute = now.minute();
  int second = now.second();
  char time[] = "hh:mm:ss";
  String time_format = now.ToString(time);
  char date[] = "DD/MM/YYYY";
  String date_format = now.ToString(date);
```

i'm doing it as in RTClib.h because IDK how to do another

can you help me?

awickert, 2025-05-11:
Hi @N3koSamuRay.

Hm! Confused on a few fronts.
1. We don't have RTClib.h here.
2. Code like this should go in a `*.cpp` file rather than a `*.h` file.
3. The function name that you're aiming for is already taken, so this would cause a conflict (see below, from [DS3231.cpp](https://github.com/NorthernWidget/DS3231/blob/master/DS3231.cpp)).
```
DateTime RTClib::now(TwoWire & _Wire) {
  _Wire.beginTransmission(CLOCK_ADDRESS);
  _Wire.write(0);	// This is the first register address (Seconds)
  			// We'll read from here on for 7 bytes: secs reg, minutes reg, hours, days, months and years.
  _Wire.endTransmission();

  _Wire.requestFrom(CLOCK_ADDRESS, 7);
  uint16_t ss = bcd2bin(_Wire.read() & 0x7F);
  uint16_t mm = bcd2bin(_Wire.read());
  uint16_t hh = bcd2bin(_Wire.read());
  _Wire.read();
  uint16_t d = bcd2bin(_Wire.read());
  uint16_t m = bcd2bin(_Wire.read());
  uint16_t y = bcd2bin(_Wire.read()) + 2000;

  return DateTime (y, m, d, hh, mm, ss);
}
```
4. There is no `ToString` method here.

Does what you are writing here relate to this library?

N3koSamuRay, 2025-05-13:
> Hi [@N3koSamuRay](https://github.com/N3koSamuRay).
>
> Hm! Confused on a few fronts.
>
> 1. We don't have RTClib.h here.
> 2. Code like this should go in a `*.cpp` file rather than a `*.h` file.
> 3. The [...]

N3koSamuRay, 2025-05-13:
THANKS SO MUCH, i didnt even think that somebody can reply me.

> Does what you are writing here relate to this library?

yes, in github of your library i saw fragment about RTCib and i thought that in DS3231 you can format time

>The Special RTClib::now() Function
>RTClib::now() is the precise, complete name for a special function that returns a DateTime object from the DS3231. Always write it >just so: RTClib::now().

>The function returns a DateTime object. To use it in your program, declare a DateTime type of variable to receive the value. For >example,
>
>DateTime currentMoment = RTClib::now();
>
>The value of currentMoment can then be accessed as either:
>
>an unsigned integer containing the number of seconds since a certain reference date, or
>distinct values for Year, Month, Day, Date, Hour, Minute, or Second.

closed 2025-10-06 by awickert